apcountryman/prototype-kicad-project-ci-cd #13: Add schematic PDF generation CI job

jaylamb, 2021-09-15:
Add schematic PDF generation CI job.

- [ ] Add schematic PDF generation CI script (`ci/generate-schematic-pdf`)
  - [ ] Supported options:
    - [ ] `--help`: See other project scripts
    - [ ] `--version`: See other project scripts
  - [ ] `function error()`: See other project scripts
  - [ ] `function abort()`: See other project scripts
  - [ ] `function display_help_text()`: See other project scripts
  - [ ] `function display_version()`: See other project scripts
  - [ ] `function main()`: See other project scripts
  - [ ] `function generate_schematic_pdf()`:
  
```bash
local schematics; mapfile -t schematics < <( git -C "$repository" ls-files '*.sch' ':!:*-sheet-*.sch' | xargs -r -d '\n' -I '{}' find "$repository/{}" ); readonly schematics

for schematic in "${schematics[@]}"; do
    if ! "$repository/utilities/KiAuto/src/eeschema_do" export -a "$schematic" "$repository/deliverables"; then
        abort
    fi
done
```

apcountryman, 2021-09-15:
The schematic PDF generation script looks good to me. Consider changing `"$repository/deliverables"` to `"$repository/schematic-pdfs"` since the directory will only contain schematic PDFs.

Please add the required `.github/workflow/ci.yml` updates to the design. I think it would be a good idea to store the generated schematic PDFs as workflow artifacts to make a PR reviewer's job easier. 7 days seems like it would be a reasonable artifact retention period. More information about storing workflow data as artifacts can be found at https://docs.github.com/en/actions/guides/storing-workflow-data-as-artifacts.

apcountryman, 2021-09-18:
A few things:
- The CI script name should probably be changed to `ci/generate-schematic-pdfs`, and the workflow job name should probably be changed to `generate-schematic-pdfs` for consistency with other verbage
- I think `Archive generated schematic PDF(s)` would be a more descriptive job step name than `Archive production artifacts`
- I think `schematic-pdfs` would be a more descriptive artifact name than `upload-schematics`
- I believe the `|` is not necessary for a single line path and thus that entry (?) could be changed to `path: schematic-pdfs`

Other than these, this design LGTM. I'll open a `type-feature` issue for implementing this new feature shortly.

closed 2021-09-18 by apcountryman