apcountryman/prototype-kicad-project-ci-cd #42: Design Test Printed Circuit Board Project

jaylamb, 2021-12-26:
## Design Demonstration Printed Circuit Board
Add a test KiCad project to the repository, consisting of a schematic and printed circuit board (PCB) layout. 

### Use Case
The purpose of the project is to test the functionality of each of the [continuous integration (CI) jobs](https://github.com/apcountryman/prototype-kicad-project-ci-cd/blob/main/.github/workflows/ci.yml) included in this repository.

### Design Description / Elements
The PCB design should be simple, as to make obvious what is being tested. As the design progresses, commits should be pushed to demonstrate that the CI pipeline will pass or fail when it is expected to. The design should test the CI pipeline's ability to detect the following deficiencies:
* [KiCad Library Convention (KLC)](https://klc.kicad.org/) compliance
* Design rules compliance (design rules check)
* Electrical rules compliance (electrical rules check)

In addition, the project should test that the CI pipeline generates deliverables for the PCB:
* PDF of the schematic
* [Traditional Bill of Materials (BOM)](https://raw.githubusercontent.com/SchrodingersGat/KiBoM/master/example/html.png)
* [Interactive HTML BOM](https://openscopeproject.org/InteractiveHtmlBomDemo/#video-demo)
* Production files (Gerbers, drills, pick-and-place)
* 3D renders of the PCB

The project should be a breakout for the [Maxim Integrated MAX6683 Temperature Sensor and System Monitor](https://www.maximintegrated.com/en/products/sensors/MAX6683.html). In addition to the part itself, the PCB design should include pin headers which will enable the PCB to be placed on a breadboard, as well as any necessary discrete components. Descriptive information should be added to the PCB's silkscreen layer, as appropriate.

### References
[MAX6683 Datasheet](https://datasheets.maximintegrated.com/en/ds/MAX6683.pdf)

apcountryman, 2021-12-26:
The overall design looks good to me, but I think it would be a good idea to go into a bit more detail now in order to avoid later revisions to the schematic and layout.
1. Which signals should be brought to the pin header(s)?
2. Should all signals go to the same header or should there be multiple headers?
3. How should the ADD pin be handled? Permanently tie it to GND, SDA, SCL or VCC? Provide a switch that allows the user to select between the options or a subset of the options? Provide a set of solder jumpers that allows the pin's connection to be configured during assembly?
4. Should the PCB provide the I2C pull-ups or should the user be expected to provide them? If the PCB provides them, should switches or solder jumpers be provide to select between several pull-up strengths?

jaylamb, 2021-12-31:
Agreed. Some more detail here will take a bunch of the mystery out of the upcoming work.

1. As the part isn't relatively complicated or large, I'd recommend bringing each of the signals (1.8Vin, 2.5Vin, 5Vin, GND, Vcc, SCL, SDA, ADD, and !ALERT) out to a pin 
2. To ease use with a breadboard, I think two 5 pin headers should be used, and the signals can just be arranged as they are on the part itself:
[image]
3. As discussed, the method used by Sparkfun for their TMP117 breakout board would be a great way to handle the configurable I2C address portion of a MAX6683 breakout:
[image]
4. As with the I2C address, the Sparkfun TMP117 breakout had an interesting idea for I2C pull-up resistors:
[image]
Recommend we either do something along these lines, or leave the resistors off, as they should (generally) be included/placed near the microcontroller (or other "master" device).

apcountryman, 2021-12-31:
1. Sounds good.
2. Sounds good.
3. Sounds good.
4. I'd like to pursue the Sparkfun I2C pull-up resistor approach since it would be convenient for the use case of simply connecting the breakout board to a dev board, but I'd be fine with removing them if including them ends up causing problems.

I'll open a `type-feature` issue for adding the schematic/layout shortly.

closed 2021-12-31 by apcountryman