apcountryman/prototype-kicad-project-ci-cd #49: Fix CI Scripts for KiCad 6 Compatibility

jaylamb, 2022-05-22:
During development of #44, a few small issues have been uncovered with the CI scripts:

- [ ] **ci/check_design_rules_compliance**
   - [ ] **Line 70**
   `cat "$repository/"*.drt` should be updated: `cat "$repository/"*.rpt`
- [ ] **ci/generate_interactive_bom**
   - [ ] **Line 62, 70**
   Instances of `.pro` should be updated to `.kicad_pro`
   - [ ] **Line 73**
   `if ! git -C ...` should be updated: `if git -C ...`
- [ ] **ci/generate-pcb-renders**
   - [ ] **Line 61, 69**
   Instances of `.pro` should be updated to `.kicad_pro`
   - [ ] **Line 72**
   `if ! git -C ...` should be updated: `if git -C ...`
- [ ] **ci/generate-production-files**
   - [ ] **Line 62, 70**
   Instances of `.pro` should be updated to `.kicad_pro`
   - [ ] **Line 73**
   `if ! git -C ...` should be updated: `if git -C ...`
- [ ] **ci/generate-schematic-pdf**
   - [ ] **Line 62, 70**
   Instances of `.pro` should be updated to `.kicad_pro`
   - [ ] **Line 71**
   `.sch` should be updated to `.kicad_sch`
   - [ ] **Line 73**
   `if ! git -C ...` should be updated: `if git -C ...`
- [ ] **ci/generate-traditional-bom**
   - [ ] **Line 62, 70**
   Instances of `.pro` should be updated to `.kicad_pro`
   - [ ] **Line 71**
   `.sch` should be updated to `.kicad_sch`
   - [ ] **Line 73**
   `if ! git -C ...` should be updated: `if git -C ...`

apcountryman, 2022-05-22:
I'll fix these shortly. Looks like `ci/check-electrical-rules-compliance` also needs a `.sch` -> `.kicad_sch` extension change so it will be added to the list of things to fix.

jaylamb, 2022-05-22:
> I'll fix these shortly. Looks like `ci/check-electrical-rules-compliance` also needs a `.sch` -> `.kicad_sch` extension change so it will be added to the list of things to fix.

Sounds good. I'll merge things in and be back at it a little later on.

PR #50 (2022-05-22): Fix CI scripts for KiCad 6 compatibility
Resolves #49 (Fix CI scripts for KiCad 6 compatibility).

This pull request:
- [x] Implements a bug fix
- [ ] Implements an enhancement to an existing feature
- [ ] Implements a new feature
- [ ] Performs a refactoring

Please mark the pull request as "Ready for review" and request a review when the pull
request is ready for a review.
If changes are requested, please discuss and/or address the review findings before
requesting a new review.

@apcountryman
commit fa51f27d (2022-05-22): Fix CI scripts for KiCad 6 compatibility (#50)

Resolves #49 (Fix CI scripts for KiCad 6 compatibility).
closed 2022-05-22 by apcountryman